daynix/UsbDk #98: UsdDk removes Audio 2.0 from CDC + Audio 2.0 device

andrew-mclachlan, 2020-12-17:
Hi there -

Hoping that someone has experience of this issue:

I am attaching a composite device which sets up CDC and Audio 2.0 input/output. I am using libUSB backed with UsbDk to connect to the CDC.

I have the device driver set-up for the CDC /w Audio 2.0 and I can communicate with it via COM serial and hear audio via Audio 2.0.

However, if I do a libusb_open on the device then UsbDk replaces the driver, from what I have read, and the Audio 2.0 is no longer available.

There is no problem on macOS or Linux. Just seems to be an issue on Windows due to UsbDk set-up.

Any advice would be greatly appreciated!!

ybendito, 2020-12-17:
When libusb uses UsbDk upon libusb_open the entire device can be accessed only by libusb via its interface with UsbDk.
You can configure libusb at initialization time to use alternate backend (probably WinUsb):
http://libusb.sourceforge.net/api-1.0/group__libusb__lib.html#gaf6ce5db28dac96b1680877a123da4fa8
http://libusb.sourceforge.net/api-1.0/group__libusb__lib.html#ga07d4ec54cf575d672ba94c72b3c0de7c

andrew-mclachlan, 2020-12-17:
Thanks. I tried omitting the USBDK option but didn't have much success with WinUSB. That's definitely a separate issue.

Thanks for the answer. We can close the issue.

closed 2020-12-18 by ybendito