daynix/UsbDk #99: The newest version 1.0.22 (downloaded from this web site) fails to install.

xiayibing, 2020-12-23:
=== Logging started: 2020/12/23  11:05:42 ===
Action 11:05:42: INSTALL. 
Action start 11:05:42: INSTALL.
Action 11:05:42: FindRelatedProducts. Searching for related applications
Action start 11:05:42: FindRelatedProducts.
Action ended 11:05:42: FindRelatedProducts. Return value 1.
Action 11:05:42: AppSearch. Searching for installed applications
Action start 11:05:42: AppSearch.
AppSearch: Property: WIN10FOUND, Signature: searchFile
Action ended 11:05:42: AppSearch. Return value 1.
Action 11:05:42: ValidateProductID. 
Action start 11:05:42: ValidateProductID.
Action ended 11:05:42: ValidateProductID. Return value 1.
Action 11:05:42: CostInitialize. Computing space requirements
Action start 11:05:42: CostInitialize.
Action ended 11:05:42: CostInitialize. Return value 1.
Action 11:05:42: FileCost. Computing space requirements
Action start 11:05:42: FileCost.
Action ended 11:05:42: FileCost. Return value 1.
Action 11:05:42: CostFinalize. Computing space requirements
Action start 11:05:42: CostFinalize.
Action ended 11:05:42: CostFinalize. Return value 1.
Action 11:05:42: ExecuteAction. 
Action start 11:05:42: ExecuteAction.
Action start 11:05:42: INSTALL.
Action start 11:05:42: FindRelatedProducts.
Action ended 11:05:42: FindRelatedProducts. Return value 0.
Action start 11:05:42: AppSearch.
Action ended 11:05:42: AppSearch. Return value 0.
Action start 11:05:42: ValidateProductID.
Action ended 11:05:42: ValidateProductID. Return value 1.
Action start 11:05:42: CostInitialize.
Action ended 11:05:42: CostInitialize. Return value 1.
Action start 11:05:42: FileCost.
Action ended 11:05:42: FileCost. Return value 1.
Action start 11:05:42: CostFinalize.
Action ended 11:05:42: CostFinalize. Return value 1.
Action start 11:05:42: InstallValidate.
Action ended 11:05:42: InstallValidate. Return value 1.
Action start 11:05:42: InstallInitialize.
Action ended 11:05:42: InstallInitialize. Return value 1.
Action start 11:05:42: ProcessComponents.
Action ended 11:05:42: ProcessComponents. Return value 1.
Action start 11:05:42: UnpublishFeatures.
Action ended 11:05:42: UnpublishFeatures. Return value 1.
Action start 11:05:42: RemoveDuplicateFiles.
Action ended 11:05:42: RemoveDuplicateFiles. Return value 1.
Action start 11:05:42: RemoveFiles.
Action ended 11:05:42: RemoveFiles. Return value 0.
Action start 11:05:42: InstallFiles.
Action ended 11:05:42: InstallFiles. Return value 1.
Action start 11:05:42: InstallDriver.
Action ended 11:05:42: InstallDriver. Return value 1.
Action start 11:05:42: DuplicateFiles.
Action ended 11:05:42: DuplicateFiles. Return value 1.
Action start 11:05:42: RegisterUser.
Action ended 11:05:42: RegisterUser. Return value 1.
Action start 11:05:42: RegisterProduct.
Action ended 11:05:42: RegisterProduct. Return value 1.
Action start 11:05:42: PublishFeatures.
Action ended 11:05:42: PublishFeatures. Return value 1.
Action start 11:05:42: PublishProduct.
Action ended 11:05:42: PublishProduct. Return value 1.
Action start 11:05:42: InstallFinalize.
CustomAction InstallDriver returned actual error code 1 (note this may not be 100% accurate if translation happened inside sandbox)
Error 1722. There is a problem with this Windows Installer package. A program run as part of the setup did not finish as expected. Contact your support personnel or package vendor.  Action InstallDriver, location: C:\Program Files\UsbDk Runtime Library\, command: C:\Program Files\UsbDk Runtime Library\\UsbDkInstHelper.exe i  
MSI (s) (88:54) [11:05:47:584]: Product: UsbDk Runtime Libraries -- Error 1722. There is a problem with this Windows Installer package. A program run as part of the setup did not finish as expected. Contact your support personnel or package vendor.  Action InstallDriver, location: C:\Program Files\UsbDk Runtime Library\, command: C:\Program Files\UsbDk Runtime Library\\UsbDkInstHelper.exe i  

Action ended 11:05:47: InstallFinalize. Return value 3.
Action ended 11:05:47: INSTALL. Return value 3.
Action ended 11:05:47: ExecuteAction. Return value 3.
Action ended 11:05:47: INSTALL. Return value 3.
=== Logging stopped: 2020/12/23  11:05:47 ===
MSI (c) (CC:60) [11:05:47:659]: Product: UsbDk Runtime Libraries -- Installation failed.

MSI (c) (CC:60) [11:05:47:659]: Windows Installer installed the product. Product Name: UsbDk Runtime Libraries. Product Version: 1.0.22. Product Language: 1033. Manufacturer: Red Hat, Inc.. Installation success or error status: 1603.

cam1170, 2021-02-19:
I also got a similar error, that there was something wrong with the MSI and it failed to install.

mrrsf, 2023-06-15:
any fix?

chjohans, 2023-10-31:
I have the exact same problem, please look into this