daynix/UsbDk #100: Dolphin emulator reports instability

Rumi-Larry, 2021-02-20:
The dolphin emulator introduced usb passtrough to support different wii usb accesories. They attempted to implement usbdk, but reported a bunch of cases where the system would not receive, crash, or hang; therefore they have removed it and settled with libusbK and winusb (the latter is incompatible with some accesories and games, and requires extra steps to work outside of Windows). 
The dolphin devs did manage to identify and report 3 bugs on libusb itself, so if the devs here engage with the dolphin devs, debugging the problems in usbdk. 
This link describes the issues: https://es.dolphin-emu.org/blog/2019/06/02/dolphin-progress-report-may-2019/?cr=es#50-10270-and-50-10344-ios-hle-fixes-and-passthrough-upgrades-by-leoetlino

ybendito, 2021-04-18:
Some improvements were done in USBDK 1.0.22 that may solve some problems in this class of applications. But we're not involved in any specific interoperability research of Dolphin emulator and UsbDk

Squall-Leonhart, 2022-11-02:
dolphin cannot use USBDK 1.0.22 without bsodding due to #115