daynix/UsbDk #101: Redirecting an interface from a composite device

k0k0c, 2021-04-16:
Hello!
I am developing a Java extension for windows application for controlling a composite USB device - MIDI controller Maschine MK3. There is a USB description here:
https://github.com/k0k0c/cabl/blob/develop/doc/hardware/maschine-mk3/MaschineMK3-Overview.md
I use HID interface 4 to receive data from buttons and send data to LED buttons, and interface 5 to send images to device screens. With interface 4, since it is HID, I had no problem, it works well with the standard driver. But interface 5 requires a custom driver to be installed, otherwise windows will not allow you to connect to the interface. I solve this problem by installing WinUSB and everything works well together as well.
My problem is that I need to keep the original functionality of the device. With a custom display driver, this feature is lost. That is, I need to switch the screen driver from the original to the custom one and vice versa on the fly.
As far as I understand, the UsbDK redirect is what I need, but I am not a C ++ programmer and, moreover, am not familiar with the development of drivers. All I could do was download the UsbDk_1.0.22_x64.msi installer and run a redirect for my device. But it worked for the entire composite device at once, I still could not figure out how to run a redirect for a specific interface of my device. I tried to use Hider, I ran UsbDkController.exe -H with parameters but to no avail, I don’t understand how the command syntax should be in my case, and in general is this what I need? I would be grateful for any advice, thanks!

k0k0c, 2021-04-17:
I discovered another problem. In my case, the redirect is launched only once, after the completion of the redirect and restarting it does not work anymore. Only reinstalling the driver helps. In this case, the command to uninstall and install driver displays an error message. Here is a commands list that in my case leads to the result described above.

```
C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r "USB\VID_17CC&PID_1600" "51F8EEE5"
Redirect USB device USB\VID_17CC&PID_1600, 51F8EEE5
Redirect of USB device failed

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -n
Enumerate USB devices
Enumeration failed

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -i
Installing UsbDk driver
UsbDk driver installation failed

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -u
Uninstalling UsbDk driver
UsbDk driver uninstall succeeded

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -n
Enumerate USB devices
Enumeration failed

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -i
Installing UsbDk driver
UsbDk driver installation succeeded but reboot is required in order to make it functional

>>> Rebot windows here.

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r "USB\VID_17CC&PID_1600" "51F8EEE5"
Redirect USB device USB\VID_17CC&PID_1600, 51F8EEE5
USB device was redirected successfully. Redirected device handle = 0000029BA4749C80
Press any key to stop redirection
Restore USB device 0000029BA4749C80
USB device redirection was stopped successfully.

C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r "USB\VID_17CC&PID_1600" "51F8EEE5"
Redirect USB device USB\VID_17CC&PID_1600, 51F8EEE5
Redirect of USB device failed
```

ybendito, 2021-04-18:
@k0k0c 
Regarding redirection of specific interface of a composite device - the UsbDk does not provide such possibility.

ybendito, 2021-04-18:
@k0k0c Regarding the problem with repetitive redirection:
Does the detach / attach of the device from the USB slot solves the problem? In order to request device's reenumeration to stop the redirection the UsbDk uses 'power cycle' request to the USB stack (actually to the USB hub).  On some platforms after such request sometimes the device does not  appear or appears after long time. We're trying to find a solution but did not succeed yet.

k0k0c, 2021-04-18:
> @k0k0c Regarding the problem with repetitive redirection:
> Does the detach / attach of the device from the USB slot solves the problem? In order to request device's reenumeration to stop the [...]

Reconnecting the device does not solve the problem. I still get the message "Redirect of USB device failed".

> @k0k0c
> Regarding redirection of specific interface of a composite device - the UsbDk does not provide such possibility.

Is it possible to add this functionality by modifying the UsbDk source code? As I wrote above, I am not a C ++ programmer and have no experience in driver development, but I am ready to find a specialist to solve this problem. How doable and difficult is this task?

ybendito, 2021-04-19:
> Is it possible to add this functionality by modifying the UsbDk source code? As I wrote above, I am not a C ++ programmer and have no experience in driver development, but I am ready to find a specialist to solve this problem. How doable and difficult is this task?
I'd say _such_ task is very untrivial especially taking in account that UsbDk is a backend of libusb and must comply with its expectations. But probably the redirection of the interface is not mandatory for your goal. IIUC you need a possibility for specific device model (or several models) to send OUT requests to specific USB endpoint (even if the existing driver for the display device also does it once in a while). So you need kind of lower filter between the existing display driver and the USB endpoint that provides such 'aside channel'. If the device side is stateless (i.e. these aside transfers do not create troubles to transfers initiated by the driver) - this seems much simpler than develop full-blown interface redirection.

alexandru-bagu, 2021-10-30:
> I discovered another problem. In my case, the redirect is launched only once, after the completion of the redirect and restarting it does not work anymore. Only reinstalling the driver helps. In this case, the command to uninstall and install driver displays an error message. Here is a commands list that in my case leads to the result described above.
> 
> UPD: After restarting windows, the redirect functionality returns without reinstalling UsbDk, but only for one redirect. When you stop the redirect and restart it in the same Windows session, an error occurs.
> 
> ```
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r "USB\VID_17CC&PID_1600" "51F8EEE5"
> Redirect USB device USB\VID_17CC&PID_1600, 51F8EEE5
> Redirect of USB device failed
> 
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -n
> Enumerate USB devices
> Enumeration failed
> 
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -i
> Installing UsbDk driver
> UsbDk driver installation failed
> 
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -u
> Uninstalling UsbDk driver
> UsbDk driver uninstall succeeded
> 
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -n
> Enumerate USB devices
> Enumeration failed
> 
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -i
> Installing UsbDk driver
> UsbDk driver installation succeeded but reboot is required in order to make it functional
> 
> >>> Rebot windows here.
> 
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r "USB\VID_17CC&PID_1600" "51F8EEE5"
> Redirect USB device USB\VID_17CC&PID_1600, 51F8EEE5
> USB device was redirected successfully. Redirected device handle = 0000029BA4749C80
> Press any key to stop redirection
> Restore USB device 0000029BA4749C80
> USB device redirection was stopped successfully.
> 
> C:\Program Files\UsbDk Runtime Library>UsbDkController.exe -r "USB\VID_17CC&PID_1600" "51F8EEE5"
> Redirect USB device USB\VID_17CC&PID_1600, 51F8EEE5
> Redirect of USB device failed
> ```

I am having the same issue although with version v1.00-22 I get failure for driver uninstall / install no matter what I do once the redirection has been closed. Only way to fix this is restarting the windows. Additionally it seems it work for one redirect no matter what device (first redirect works fine, nothing else work afterwards even if its another device).