daynix/UsbDk #103: usbkdcontroller -n fails

xiangjiang-cn, 2021-09-14:
When the USB root controller driver of win7 is version 3.1, usbkdcontroller -n fails to enumerate USB devices. What should I do with this?

ybendito, 2021-09-14:
Please collect logs as follows: (all operations from admin console in %programfiles%\usbdk runtime library
Uninstall UsbDk driver: usbdkcontroller -u
Copy these files to %programfiles%\usbdk runtime library (attached usbdktrace.bat + tracefmt.exe, assuming you have 64-bit machine) 
[usbdk.trace.zip](https://github.com/daynix/UsbDk/files/7159785/usbdk.trace.zip)
run usbdktrace.bat (separate admin console)
Install UsdbDk driver: usbdkcontroller -i
Wait for the end of the installation
Hit enter in the trace console
There should be parsed log file, please attach it

ybendito, 2021-09-14:
Which version of USBDK you use? The log file is not parsed properly. Please attach the ETL from the same directory.

xiangjiang-cn, 2021-09-14:
> 
> 
> Which version of USBDK you use? The log file is not parsed properly. Please attach the ETL from the same directory.

The version of USBDK is 1.0.22
[usbdk.zip](https://github.com/daynix/UsbDk/files/7161489/usbdk.zip)

ybendito, 2021-09-14:
https://github.com/daynix/UsbDk/blob/master/UsbDk/FilterDevice.cpp#L605
Need to add IUSB3\ROOT_HUB31

xiangjiang-cn, 2021-09-15:
> https://github.com/daynix/UsbDk/blob/master/UsbDk/FilterDevice.cpp#L605
> Need to add IUSB3\ROOT_HUB31

Ok, thank you very much!