daynix/UsbDk #105: USB redirection stuck

kvojacheck, 2021-11-09:
If a client application which is holding an open handle to a redirected USB port is force-terminated by an external program (e.g. *task manager*), the redirected port becomes unresponsive, and cannot be used until system reboot.

Since UsbDk can receive device open/close requests from multiple processes concurrently, it tries to close the USB redirection only if the close was requested by the same process that started the redirection.
The problem is that the WDF file close callback is not guaranteed to be called from the same process that is opened the file handle, but from an *arbitrary process context* (see reference [here](https://www.osr.com/nt-insider/2017-issue2/handling-cleanup-close-cancel-wdf-driver/)), though in practice, it seems to be mostly called from the initiator process, making it difficult to trigger this bug.

One solution could be switching from `PsGetCurrentProcessId()` to `WdfFileObjectGetInitiatorProcessId()`, which seems to return the expected PID in all cases. Unfortunately, this method is not available until KMDF 1.21, meaning that UsbDk wouldn't work on versions of Windows earlier than *Windows 10, version 1703*.

Therefore, to keep compatibility, I would suggest adding a reference count to the device, so it could be ensured that the redirection is removed when the last reference is closed. This would improve the current situation without known functionality/compatibility losses.

mxk, 2022-10-05:
Just ran into this issue. Does the PR require more work?

starsunyzl, 2023-02-20:
Facing the same issue. It's quite often need to force terminate the process in the development process, since then, StartRedirect() will never succeed again unless reboot the system.