daynix/UsbDk #107: error on install msi

Marthianfred, 2021-11-22:
msi stuck on error 

[image]

any help?

ybendito, 2021-11-23:
Operating system? UsbDk version?

ybendito, 2021-11-30:
Let's try to get logs from the MSI
create c:\log directory
from admin command line
msiexec /i <msifile> /L*V "C:\log\usbdk.log"
let's look what we can see in usbdk.log

NO111ONE, 2021-12-02:
> Let's try to get logs from the MSI create c:\log directory from admin command line msiexec /i /L*V "C:\log\usbdk.log" let's look what we can see in usbdk.log

Here's a first-install (that magically started working) and a post-reboot.
"UsbDkController" is still not a command, though.
[postrebootusbdk.log](https://github.com/daynix/UsbDk/files/7644898/postrebootusbdk.log)
[usbdk.log](https://github.com/daynix/UsbDk/files/7644899/usbdk.log)

Hallohannes123, 2023-04-30:
I have this same issue. I installed it before, wanted to reinstall it and now the installer is stuck.
Any solutions?

YanVugenfirer, 2023-04-30:
> I have this same issue. I installed it before, wanted to reinstall it and now the installer is stuck.
> Any solutions?

Please provide the log and information regarding the OS.
```
create c:\log directory
from admin command line
msiexec /i /L*V "C:\log\usbdk.log"
let's look what we can see in usbdk.log
```

Hallohannes123, 2023-04-30:
> Please provide the log and information regarding the OS.
I am on Windows 11 Beta Insider build.
After installing my usb devices stopped working, and safe mode etc. didn't fix it. I am now resetting my pc, I'll try again afterwards and provide logs if it fails again

Hallohannes123, 2023-04-30:
After resetting windows it installed without any issues. Not sure what happened
[usbdk.log](https://github.com/daynix/UsbDk/files/11361283/usbdk.log)
I created this log before USB stopped working entirely.

Hallohannes123, 2023-04-30:
In the logs it mentions
```
MSI (s) (5C:AC) [11:16:21:319]: SECREPAIR: New Hash Database creation complete.
MSI (s) (5C:AC) [11:16:21:328]: SECREPAIR: Failed to open the file:C:\Program for computing its hash. Error:2
```
This seems odd. "C:\Program" sounds like some space key confusion?

Jevon-V, 2023-06-30:
I also have this problem, but in most cases, the progress bar is not moved, and then the system cannot be closed properly after executing the cancel installation (windows7 Professional sp1, usbdk1.0.22)
[usbdk.log](https://github.com/daynix/UsbDk/files/11914480/usbdk.log)

ybendito, 2023-07-01:
@Jevon-V For Win7: do you have https://learn.microsoft.com/en-us/security-updates/SecurityAdvisories/2015/3033929?redirectedfrom=MSDN installed?

Jevon-V, 2023-07-01:
yep, it is already installed

ybendito, 2023-07-01:
@Jevon-V 1.0.21 behaves the same?

Jevon-V, 2023-07-04:
ye, the effect is the same

ybendito, 2023-07-04:
@Jevon-V I suggest to try manual install, see https://github.com/daynix/UsbDk/wiki/Troubleshooting-UsbDk-installation

Jevon-V, 2023-07-04:
I tried to unpack the msi and manually execute the command to install it, but the effect was still the same. After canceling, the restart could not be performed normally

Jevon-V, 2023-07-04:
@ybendito According to the code debugging, installation progress is stuck in m_wdfCoinstaller PostDeviceInstall (infFilePath);

ybendito, 2023-07-04:
@Jevon-V which WDF you have on your Win7 (check the version of drivers\wdf01000.sys)?

Jevon-V, 2023-07-04:
@ybendito 1.11.9200.16648

ybendito, 2023-07-04:
@Jevon-V Interesting. I'd expect the WDF installation should just check the version and exit. Are you able to build the UsbDkHelper.dll? If yes, I'll send you a patch to try.

Jevon-V, 2023-07-04:
@ybendito yep, I can compile

ybendito, 2023-07-04:
@Jevon-V 
[0001-UsbDkHelper-avoid-WDF-installation.patch](https://github.com/daynix/UsbDk/files/11947701/0001-UsbDkHelper-avoid-WDF-installation.patch)
Apply this patch and build the UsbDkHelper for your system (32 or 64), just replace it under \Program Files\UsbDk ... and retry the install

Jevon-V, 2023-07-04:
\UsbDk Runtime Library>.\UsbDkController.exe -i
Installing UsbDk driver
UsbDk_InstallDriver
install failure
UsbDk driver installation failed

ybendito, 2023-07-04:
> UsbDk Runtime Library>.\UsbDkController.exe -i
> Installing UsbDk driver
> UsbDk_InstallDriver
> install failure
> UsbDk driver installation failed

Jevon-V, 2023-07-04:
@ybendito oh, sry, I commented before that the reason why m_wdfCoinstaller.PostDeviceInstall(infFilePath) did not block error was that there was security software left on. I just used your patch to compile and still blocked. I found that the UpperFilters of registry System\\CurrentControlSet\\Control\\Class\\{36FC9E60-C465-11CF-8056-444553540000} are usbfilter(AMD USB Filter Driver). Whether to uninstall it to install the usbdk.

ybendito, 2023-07-04:
@Jevon-V In general there is no problem of coexistence among several filters.
Let's try rebootRequired = true (see the note inside my patch)

Jevon-V, 2023-07-05:
@ybendito 
\UsbDk Runtime Library>.\UsbDkController.exe -i
Installing UsbDk driver
UsbDk_InstallDriver
UsbDk driver installation succeeded but reboot is required in order to make it functional

ybendito, 2023-07-05:
@Jevon-V Sure, (see the code) in this case the installer just creates the service and registers it in the registry, but does not renumerate the devices. Reenumeration and inclusion of UsbDk into each USB device's stack happens on reboot.
If you reboot the machine after that - does the UdnDk works? (for example UsbDkController -n)

Jevon-V, 2023-07-05:
yep, usbdk can be used normally, ty so much

ybendito, 2023-07-05:
@Jevon-V 
OK, so it looks li9ke there are 2 problems on your Win7:
- wdf update (probably)
- explicit reenumeration of the devices
Can you attach the output of "UsbDkController -n"?
Would you mind to help in understanding what is the problem 2?
If yes:
- Does 'UsbDkController -u' works now?
- If you rebuild the UsbDkHelper.dll with  'rebootRequired = true', does the 'UsbDkController -i' work now?

Jevon-V, 2023-07-05:
"UsbdkController -u" is always working properly,  and "UsbdkController -i" can also run normally after rebuild UsbDkHelper.dll with rebootRequired = true.

the UsbContrtoller -n output:
Enumerate USB devices
Found 3 USB devices:
0. FilterID: 2, Port: 1, ID: 0951:1666, Configs: 1, Speed: 3
        USB\VID_0951&PID_1666 E0D55E55CC73E641292C229D
Descriptor for configuration #0: size 32
1. FilterID: 3, Port: 4, ID: 17ef:608d, Configs: 1, Speed: 2
        USB\VID_17EF&PID_608D 4
Descriptor for configuration #0: size 34
2. FilterID: 3, Port: 5, ID: 17ef:6099, Configs: 1, Speed: 2
        USB\VID_17EF&PID_6099 5
Descriptor for configuration #0: size 34

ybendito, 2023-07-05:
@Jevon-V 
Sorry, my mistake
If you rebuild the UsbDkHelper.dll with 'rebootRequired = false', does the 'UsbDkController -i' work now?

Jevon-V, 2023-07-05:
@ybendito No, it'll keep blocking

ybendito, 2023-07-05:
@Jevon-V Blocking or failing?

Jevon-V, 2023-07-05:
It's mostly blocking, sometimes errors occur, and when the installation goes wrong, performing the uninstallation will also report errors

weskerty, 2024-04-26:
[image]
same?