daynix/UsbDk #108: redirect failed on windows7

xiangjiang-cn, 2021-12-10:
This issue is similar to issue 26. Driver file operation error. DeviceIoControl failed (The device is not connected. Error code = 1167) Redirect of USB device failed

xiangjiang-cn, 2021-12-10:
usbdk version is 1.0.22

ybendito, 2021-12-10:
Is it similar to https://github.com/daynix/UsbDk/issues/103?

xiangjiang-cn, 2021-12-10:
no, it is similar to #26

ybendito, 2021-12-10:
Please collect the log as explained in https://github.com/daynix/UsbDk/issues/103
Upload the log and etl file (to be on the safe side)

xiangjiang-cn, 2021-12-13:
> Please collect the log as explained in #103 Upload the log and etl file (to be on the safe side)
[usb_log.zip](https://github.com/daynix/UsbDk/files/7700316/usb_log.zip)

ybendito, 2021-12-13:
Please upload the etl (the log is not parsed correctly, probably due to win7)

xiangjiang-cn, 2021-12-14:
> 
[usb_log-20211214.zip](https://github.com/daynix/UsbDk/files/7708071/usb_log-20211214.zip)
The log has been uploaded, but now I cannot reproduce the problem.

ybendito, 2021-12-17:
@xiangjiang-cn In the log I do not see any problem, but also I do not see any attempt to redirect USB device.

xiangjiang-cn, 2022-01-11:
[20220111.zip](https://github.com/daynix/UsbDk/files/7843023/20220111.zip)
[20220111-1.zip](https://github.com/daynix/UsbDk/files/7843024/20220111-1.zip)