daynix/UsbDk #110: Blue Screen of Death After Installing on WINDOWS 10

ryujincoder, 2022-01-05:
I needed to install usbdk because I needed to run mtk-client witch required it. After I installed the usbdk version 1.00-22 in boot camp, I got a blue screen that said, INACCESIBLE_BOOT_DEVICE. I could NOT boot into safe mode or, anything else and I was bootlooping non stop. I dont know if this is something in boot camp but it was non funtional

ybendito, 2022-01-05:
Can you please describe what is 'boot camp' in this context? What is the boot drive of the system? (is it USB drive?)
Can you run usbview application from WDK and save/export the usb tree to a text file?

ryujincoder, 2022-01-05:
The "Boot Camp" thing I am talking about is the Boot Camp Assisrtant Utility that is bused to install Windows 10 on a mac(because I do not own a real pc)
[image]

ryujincoder, 2022-01-05:
And it has its own partition on the macs hard drive

ybendito, 2022-01-05:
Can you run usbview application from WDK on your installed Windows VM and save/export the usb tree to a text file?

ryujincoder, 2022-01-05:
Yes

ryujincoder, 2022-01-05:
But not on bootcamp

AladdinSy, 2022-05-21:
Hello, The same issue here.
I have installed win10 x64 on USB flash drive and getting INACCESIBLE_BOOT_DEVICE after install Usbdk 1.0.2-22.

owenstake, 2023-01-22:
you need to uninstall usbdk to recover your win10

parkerlreed, 2023-04-20:
Yep same here. Installing USBDK prevents a USB install of Windows from booting with the inaccessible boot device error. 

Windows 11

Squall-Leonhart, 2025-01-15:
preventing hdd installs too on windows 11, its not just usb affected.

owenstake, 2025-05-07:
> ~preventing hdd installs too on windows 11, its not just usb affected.~
> 
> false alert, this was due to a bad removal of a lower filter.
> 
> The problem with USBDK on USB OS installs is that it breaks the enumeration of USB 3.0 ROOT Hubs when its driver is loaded onto the controller.

So how to deal with it?

ybendito, 2025-05-08:
@owenstake ChatGPT recommends following:
Mark usbdk.sys as starting on boot (Start=0) in UsbDK service registry
Copy the usbdk.sys binary to %windir%\system32\drivers
Change the path in UsbDK service registry to be %systemroot%\system32\drivers

yashikada, 2025-06-08:
> [@owenstake](https://github.com/owenstake) ChatGPT recommends following: Mark usbdk.sys as starting on boot (Start=0) in UsbDK service registry Copy the usbdk.sys binary to %windir%\system32\drivers Change the path in UsbDK service registry to be %systemroot%\system32\drivers\usbdk.sys

thank you @ybendito and ChatGPT for the suggestion but doesn't works. The driver is already installed in %systemroot%\system32\drivers\usbdk.sys, start=0 don't improve the result.
I tested also on win11 without success.

Squall-Leonhart, 2025-06-08:
of course it doesn't work, the device gets removed when the handover from bios to OS occurs as there are no devices listed below the root hub when usbdk is loaded on it.