daynix/UsbDk #111: Problem with AbortPipe

AiCabal, 2022-04-20:
I want to cancel transfer and tried to do it with two way: CancelIoEx (like libusb) and AbortPipe.
AbortPipe doesn't work at all while CancelIoEx works. AbortPipe  return 0 value and all bytes are transfered. But also for some users doesn't work CancelIoEx it either transmits all bytes and return 1 or does not transmit at all and anyway return 1.

ybendito, 2023-07-11:
@AiCabal Which kind of transfer you try to cancel by aborting the pipe?

P5-2005, 2023-08-10:
> @AiCabal Which kind of transfer you try to cancel by aborting the pipe?

seem like this :
https://github.com/mineek/openra1n/blob/4595a5333e4134ade77b43fb2259e880b85801ee/openra1n.c#L785C50-L785C50

ybendito, 2023-08-10:
@P5-2005 Disclaimer: If I'm not mistaken:
Aborting of the control pipe does not do anything and this does not depend on UsbDk. Control transaction has its own timeout of 5 seconds. You can cancel the request, but the transaction will be finished or stall condition will be automatically cleared. Because this is control pipe. On the rest of pipes the stall need to be cleared manually, for that there is a pipe abort. Details should be checked in the USB spec.

P5-2005, 2023-10-30:
the issue i get on my side, its usbdk doesnt do redirection on usb 2.0 with two controller, i dont have now the debug for such case because i have now only pc with usb 3.0 and one controller host so its fine
wonder if there is workaround to choose or precise which controller must forward into
[usb2.0.txt](https://github.com/daynix/UsbDk/files/13208795/usb2.0.txt)

P5-2005, 2023-10-30:
> @P5-2005 Disclaimer: If I'm not mistaken: Aborting of the control pipe does not do anything and this does not depend on UsbDk. Control transaction has its own timeout of 5 seconds. You can cancel [...]

also that timeout was add if not using usbdk, but usbdk as backend is better than this timeout implemented with libusb(seem not enough)

P5-2005, 2023-10-30:
i found an old trace debug on pc with two usb host controller and usb2.0 only :
[usbdkmanyroothub2.0.txt](https://github.com/daynix/UsbDk/files/13208908/usbdkmanyroothub2.0.txt)