daynix/UsbDk #112: Failed to compile on new setup

pcorbineau, 2022-07-29:
I tried to compile project on recent setup with Visual Studio 2022 and 2019 with WDK 10 installed but it failed.

I got some trouble about headers in stdafx.h like 
```
#include <ntifs.h>
#include <wdf.h>
```
Project can't found those headers. If i add it manually to include directories, it failed later on redefinition of objects.

Did I miss something ?

I even try on other setup to install VS2015 but got some errors too.

YanVugenfirer, 2022-07-31:
Hi,

USBDK wasn't adopted to VS2019 and up. Usually when MS switches to new version there are some adaptations needed in projects files.

Try to follow https://github.com/virtio-win/kvm-guest-drivers-windows/wiki/Building-the-drivers-using-Visual-Studio-2017#obsolete-building-windows-8-windows-2012-drivers-and-up while installing your environment. Please post errors that you will get.
 
Best regards,
Yan.

YanVugenfirer, 2022-08-02:
https://stackoverflow.com/questions/39601108/fatal-link-error-and-can-not-find-cfg-support-v1-lib

I think it is a question of SDK\DDK compatibility. 
Just in case, recreate development env and follow also order of the instructions while installing the components.

pcorbineau, 2022-08-03:
Ok after uninstalling all SDK/WDK on my pc, then reinstalling SDK10 and WDK10 it works.
I am now be able to work on it.
I am investigating this issue (https://github.com/daynix/UsbDk/issues/99 & https://github.com/daynix/UsbDk/issues/107) that cause many problems for our customers.
If you have any tips for it 🙏

YanVugenfirer, 2022-08-03:
Try to get the logs (#107 has instructions how to gather installation logs)

YanVugenfirer, 2022-08-07:
> Ok after uninstalling all SDK/WDK on my pc, then reinstalling SDK10 and WDK10 it works.
> I am now be able to work on it.
> I am investigating this issue (#99 & #107) that cause many problems for our customers.
> If you have any tips for it 🙏

Closing based on this comment

closed 2022-08-07 by YanVugenfirer