daynix/UsbDk #116: UsbDK Installation failing- There is a problem with this Windows Installer package etc

lgexalter, 2023-01-12:
When trying to install UsbDK in Windows 10 I am getting this erorr: **There is a problem with this Windows Installer package. A program run as part of the setup did not finish as expected. Contact your support personnel or package vendor.**
[image]

I previously had UsbDK installed and after uninstalling I am getting this error. I tried redownloading the installer, running as admin, from elevated Command Prompt, and in safe mode. [Here ](https://drive.google.com/file/d/1XXeZjqJu9WjR_ujjHtqaow-YXbhVagaI/view?usp=drivesdk) are my logs

lgexalter, 2023-01-30:
I am getting this from Windows Event Viewer. `Product: UsbDk Runtime Libraries -- Error 1722. There is a problem with this Windows Installer package. A program run as part of the setup did not finish as expected. Contact your support personnel or package vendor.  Action InstallDriver, location: C:\Program Files\UsbDk Runtime Library\, command: C:\Program Files\UsbDk Runtime Library\\UsbDkInstHelper.exe i `
It seems like the issue is with UsbDkInstHelper.exe

ybendito, 2023-02-01:
Please try following:
Unpack the MSI to some subdirectory, like this
cd c:\users\user\downloads
mkdir 122
msiexe /a UsbDk_1.0.22_x64.msi /qb TARGETDIR=%cd%\122
open administrator command line
go to the directory where the usbdkcontroller.exe is placed:
cd c:\users\user\downloads\122\USB Runtime Library etc
usbdkcontroller.exe -i

It will try to install the USBDK driver.
To uninstall it you can use 
usbdkcontroller.exe -u (also from the admin command line)

lgexalter, 2023-02-01:
@ybendito 
Thanks for the reply.
Assuming you meant msiexec /a UsbDk_1.0.22_x64.msi /qb TARGETDIR=%cd%\122

I followed the instructions but it still failed.
```
Installing UsbDk driver
UsbDk driver installation failed
```

ybendito, 2023-02-01:
@lgexalter 
I suggest to download
https://learn.microsoft.com/en-us/sysinternals/downloads/debugview
Run it (as an admin) before "usbdkcontroller -i", select "Capture->Win32 and Global Win32", then run "usbdkcontroller -i"
In case of error there should be some debugging output visible in DebugView

lgexalter, 2023-02-01:
@ybendito 
```
00000001	0.00000000	[2264] UsbDkInstaller exception: CopyFile from C:\Users\Shani\Documents\122\UsbDk Runtime Library\UsbDk Runtime Library\UsbDk.sys to C:\Windows\System32\Drivers\UsbDk.sys failed. (The file exists.  Error code = 80)
```
Tried deleting and getting file in use error. But at least now I know what the error is.
Thanks!

ybendito, 2023-02-01:
Try "Usbdkcontroller.exe -u", then reboot, then "Usbdkcontroller.exe -i"

lgexalter, 2023-02-01:
@ybendito 
It worked! Thanks!!!