daynix/UsbDk #117: usbdk can't work with some USB serial device(eg 076b:23c3)  twice

crazyboy01crazy, 2023-02-02:
when I use ”usbDkController -r“ to  redir it first time, It's OK. but when I redir it again after stop the first time redir, Faild!
``` 
C:\Program Files\UsbDk Runtime Library>UsbDkController -r "USB\VID_067B&PID_23C3" BDEHf103Y23
Redirect USB device USB\VID_067B&PID_23C3, BDEHf103Y23
USB device was redirected successfully. Redirected device handle = 0000028AE8A41A10
Press any key to stop redirection
Restore USB device 0000028AE8A41A10
USB device redirection was stopped successfully.

C:\Program Files\UsbDk Runtime Library>UsbDkController -r "USB\VID_067B&PID_23C3" BDEHf103Y23
Redirect USB device USB\VID_067B&PID_23C3, BDEHf103Y23
Redirect of USB device failed
```

crazyboy01crazy, 2023-02-02:
PS：on windows10

ybendito, 2023-07-11:
@crazyboy01crazy If you replug the device - does it solve the problem?

crazyboy01crazy, 2023-11-02:
> @crazyboy01crazy If you replug the device - does it solve the problem?

Replug it can resolve the problem as I remember.

But , I need to use it without replug it. I try to find it why
`NTSTATUS CUsbDkControlDevice::AddRedirect(const USB_DK_DEVICE_ID &DeviceId, HANDLE RequestorProcess, PHANDLE RedirectorDevice)
{
    CUsbDkRedirection *Redirection;
    auto addRes = AddRedirectionToSet(DeviceId, &Redirection);
    if (!NT_SUCCESS(addRes))
    {
        return addRes;
    }
    Redirection->AddRef();
    CObjHolder<CUsbDkRedirection, CRefCountingDeleter> dereferencer(Redirection);

    TraceEvents(TRACE_LEVEL_INFORMATION, TRACE_CONTROLDEVICE, "%!FUNC! Success. New redirections list:");
    m_Redirections.Dump();

    auto resetRes = ResetUsbDevice(DeviceId, true);
    if (!NT_SUCCESS(resetRes))
    {
        TraceEvents(TRACE_LEVEL_ERROR, TRACE_CONTROLDEVICE, "%!FUNC! Reset after start redirection failed. %!STATUS!", resetRes);
        AddRedirectRollBack(DeviceId, false);
        return resetRes;
    }

    auto waitRes = Redirection->WaitForAttachment();
    if ((waitRes == STATUS_TIMEOUT) || !NT_SUCCESS(waitRes))
    {
        TraceEvents(TRACE_LEVEL_ERROR, TRACE_CONTROLDEVICE, "%!FUNC! Wait for redirector attachment failed. %!STATUS!", waitRes);
        AddRedirectRollBack(DeviceId, true);
        return (waitRes == STATUS_TIMEOUT) ? STATUS_DEVICE_NOT_CONNECTED : waitRes;
    }

    auto status = Redirection->CreateRedirectorHandle(RequestorProcess, RedirectorDevice);
    if (!NT_SUCCESS(status))
    {
        TraceEvents(TRACE_LEVEL_ERROR, TRACE_CONTROLDEVICE, "%!FUNC! CreateRedirectorHandle() failed. %!STATUS!", status);
        AddRedirectRollBack(DeviceId, true);
        return STATUS_DEVICE_NOT_CONNECTED;
    }

    return STATUS_SUCCESS;
}`

After debugging, I find that the corresponding event such as WaitForAttachment cannot be detected after executing the ResetUsbDevice operation