daynix/UsbDk #120: USB not working after install/uninstall

Hallohannes123, 2023-04-30:
I installed UsbDk and my mouse and keyboard stopped working. booting without driver signature didn't fix it. In Troubleshooting i installed and then uninstalled it with UsbDkController.exe -u, but my keyboard still won't work.
How do I fix this?

Hallohannes123, 2023-04-30:
Resetting windows "fixed" it and afterwards the installer ran without issues.

gwgill, 2023-10-25:
You can disable UsbDk from the recovery command prompt with:
REG ADD HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\UsbDk /f /v Start /t REG_DWORD /d 4
In the worst case you might have to start regedit and load the c:\Windows\system32\config\SYSTEM hive to set this value.