daynix/UsbDk #123: UsbDk automatic disable usb-device

d1ret, 2023-08-23:
When I make a request, the device is disconnected from the computer, "URB_FUNCTION_ABORT_PIPE" is called, followed by an error. Is it normal or not?

ybendito, 2023-08-23:
We do not understand the question. Can you please describe your experience in details? What does it mean: "When I make a request"? Where you see that "URB_FUNCTION_ABORT_PIPE is called"? etc

d1ret, 2023-08-23:
I have code 
```
import usb.core
import usb.util
from usb.backend import libusb1
from ctypes import c_void_p, c_int
import time

backend = usb.backend.libusb1.get_backend(find_library=lambda x: "libusb-1.0.dll")
backend.lib.libusb_set_option.argtypes = [c_void_p, c_int]
backend.lib.libusb_set_option(backend.ctx, 1)
device = usb.core.find(idVendor=0x25A7, idProduct=0xfa7c, backend=backend)
print(device)

def get_battery():
    req =  device.ctrl_transfer(bmRequestType=0x21, bRequest=0x09, wValue=0x0308, data_or_wLength=bytes(
        b"\x08\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x49"), wIndex=1)
    usb.util.dispose_resources(device)
    time.sleep(4)
    result = device.ctrl_transfer(bmRequestType=0xa1, bRequest=0x01, wValue=0x0308, data_or_wLength=40, wIndex=1)

    usb.util.dispose_resources(device)
    usb.util.release_interface(device, 0)
    print(f"Message received from the mouse: {list(result)}")

if __name__ == '__main__':
     get_battery()

```

When usbdk send request 
 ```
device.ctrl_transfer(bmRequestType=0x21, bRequest=0x09, wValue=0x0308, data_or_wLength=data, wIndex=1)"
```
my usb device disconnects from my computer, and connect after request

ybendito, 2023-08-23:
What you expect this request will do?

d1ret, 2023-08-23:
I send a request to a usb device to get a battery charge, but I need it to continue working during the request

ybendito, 2023-08-28:
It looks like the device is not happy with the request and does the internal reset (disconnects from the bus and reconnects again).