daynix/UsbDk #127: UsbDk cause win10 system in vhdx fail to startup

owenstake, 2023-12-20:
My win10 system is installed in vhdx file in usb disk. After install then UsbDk driver, this win10 system is fail to startup. After rescue by remove the UsbDk driver, win10 finally startup.

ybendito, 2023-12-20:
Thank you for the report. I suspect that because the usbdk is defined as system service and not as as boot service. I can't recommend any workaround right now except of running it from the HDD. Please let the issue open, if we find some workaround we update the issue.

ybendito, 2023-12-20:
@owenstake Did the system ask you to reboot when you tried to istall the USBDK?

owenstake, 2023-12-21:
> @owenstake Did the system ask you to reboot when you tried to istall the USBDK?

It did.

ybendito, 2023-12-22:
@owenstake 
Are you able to build your own usbdk driver? If so, may be I can suggest some change.

owenstake, 2023-12-24:
> @owenstake Are you able to build your own usbdk driver? If so, may be I can suggest some change.

I can try it.

owenstake, 2023-12-24:
> @owenstake Are you able to build your own usbdk driver? If so, may be I can suggest some change.

@ybendito  I can try it.

owenstake, 2024-01-09:
Usbdk driver will reload all usb device at win10 system boot time. Unfortunately my win10 system is in the usb device.

ybendito, 2024-01-10:
@owenstake I suggest to try one thing:
After UsbDK installation do not reboot.
Enter regedit HKLM->System->CurrentControlSet->Service-UsbDK and edit the 'Start' and create the 'Group' like that:
[image]
Then reboot
Does it help?

owenstake, 2024-01-25:
> @owenstake I suggest to try one thing: After UsbDK installation do not reboot. Enter regedit HKLM->System->CurrentControlSet->Service->UsbDK, edit the 'Start' and create the 'Group' like that: ![Capture](https://private-user-images.githubusercontent.com/23101041/295741106-e71564fb-fb83-4e52-a9e8-f737d67ea9ce.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYxNDU2NjIsIm5iZiI6MTcwNjE0NTM2MiwicGF0aCI6Ii8yMzEwMTA0MS8yOTU3NDExMDYtZTcxNTY0ZmItZmI4My00ZTUyLWE5ZTgtZjczN2Q2N2VhOWNlLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI1VDAxMTYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM2ZjQ1NTYxZDNmMjIwY2IzMzI1YWUyNDg0ZTM4YjBkMzNiYjZhZmYwMWFjNTRkMWZmNWY0YmE2ZDQxYzUxYTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.K5klB_14v299ICRC82NWZcotD0aJX2zVmiW4loEkp80) Then reboot Does it help?

I tried but failed.

ybendito, 2024-01-27:
@owenstake As far as I understand, your system is in fact something like Windows To Go (WTG).
If so, probably it should follow the same recommendations as MSFT sets for WTG, one of them - do not use non-Microsoft core USB drivers, see for ex. [here](https://learn.microsoft.com/en-us/windows/deployment/planning/best-practice-recommendations-for-windows-to-go)
The plan regarding this issue - analyze it and decide whether we're able to support it. If not - whether we're able to prevent it. I'm not sure whether even preventing is possible: you can install the UsbDk when the VHDX file is on HDD and then move the VHDX to USB and boot from it.
I do not close the issue, leaving for further analysis.

owenstake, 2024-01-29:
> @owenstake As far as I understand, your system is in fact something like Windows To Go (WTG). If so, probably it should follow the same recommendations as MSFT sets for WTG, one of them - do not use [...]

Thx for your attention.