daynix/UsbDk #128: When launching MSI Installer, installer fails

DaveTmire85, 2024-03-30:
First things first, I'm running Win11 Pro.

When I launch the installer (x64 or x86), it flashes for about a second, and then I get nothing. I have no way to know if it's working, or if it failed, or what's going on. Is this a bug or a feature?

ybendito, 2024-04-02:
Look at [wiki](https://github.com/daynix/UsbDk/wiki/Troubleshooting-UsbDk-installation) and try to collect installation log.

DaveTmire85, 2024-04-02:
After some digging, I found that though the installer closes almost as soon as it is launched, it does work as expected. I confirmed this using the "reg query HKLM\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000} /v UpperFilters" and "sc query usbdk" commands. Closing the issue.

closed 2024-04-02 by DaveTmire85