daynix/UsbDk #129: Install grinds entire system to a halt

anthonyjr2, 2024-04-21:
Windows 10 22H2 19045.4046

I installed this to use with a cheap macro pad I received, and it immediately started causing issues. Would constantly get DRIVER_POWER_STATE_FAILURE during reboots and startup that were linked to the usbdk driver. Would also slow the PC to a halt during operation and prevent me from being able to install/uninstall basically anything. Also created very weird audio issues like a delay for 10-15 seconds as well as making USB devices stop working at random intervals.

Took a lot of effort to remove in safe mode but eventually did so after force starting the Windows Installer service. Issues immediately cleared up.

Not sure if it is a new Windows update causing issues or what.

ybendito, 2024-04-21:
@anthonyjr2 
First recommendation is to try 1.00.21, the flow that on **some devices** may cause DRIVER_POWER_STATE_FAILURE  was introduced in 1.00.22

anthonyjr2, 2024-04-22:
Ah okay, I didn't see an issue that described the problem on 1.00.22 so didn't try the earlier versions. If I end up trying again I will attempt 1.00.21 but it was such a headache the last time I am going to wait a while, lol. Is there any plan to release a new version fixing the regression?

ybendito, 2024-04-22:
@anthonyjr2

- Does 1.00.22 cause any issue when said specific device is not connected? If so, I wouldn't recommend also 1.00.21. 
- Possible problem with this specific device is that it is constantly in low power mode and the UsbDk can't attach to it. 1.0.22 tries to move the device to working mode but it appears that on some devices this causes problems. **Possible** alternative is manually turn off the 'USB selective suspend', on laptops this increases power consumption
- At the moment we do not have plans for next release until we have resources for that
- The project is open-source, can be built by anyone with Visual studio and WDK, some simple intervention in build scripts might be required, packaging into MSI is not mandatory, the problematic commit is  [this one](https://github.com/daynix/UsbDk/commit/c565bd513ef3c15ca49942a7e8ef964787b337a9)

anthonyjr2, 2024-04-30:
It did not seem to be device-dependent. Simply having UsbDk installed would cause the slowdown, even after unplugging all USB devices completely. Every USB port would have issues as well, for example if I tried to plug something in, it wouldn't be recognized at all until I did a reboot. Which wasn't a very good fix because half the time the system would refuse to start with the BSOD I mentioned. The computer would also refuse to reboot from within Windows. It would hang for 5-10 minutes on the "Restarting" screen then crash with the same BSOD.

I don't plan to use it again for the time being but just figured it would be helpful to put this issue out there.

ybendito, 2024-05-01:
@anthonyjr2 UsbDk is not expected to make any negative impact at all until **really** used for device redirection or hiding. Just as an example, I have it installed for many years on main development machines over several Win10 upgrages, currently with 19045. If you have some %windir%\memory.dmp from 1.0.21 of attempt to reboot, it would be good to zip it and share with us somehow for study.

GREYALLDAY, 2024-08-15:
Hello, I would also like to add that I have had very similar issues. This package was installed as part of the edgetpu windows driver and during the installation process it froze on installing usbdk. After about an hour I finally rebooted and was also given a BSOD with DRIVER_POWER_STATE_FAILURE. After several reboots I am able to get to my windows login screen but none of my USB devices are being detected now. I've gotten into into recovery mode and subsequently tried to boot into safe mode but it hangs on my mobos splash screen. Sort of stuck atm, any tips would be appreciated. Thanks!

anthonyjr2, 2024-08-16:
@GREYALLDAY Try disconnecting all USB devices and seeing if you can boot into safe mode. Try the different types to see if it makes a difference (with networking, with command prompt, etc.). Once you get into safe mode you can force start the Windows Installer service and from there should be able to uninstall usbdk. I haven't attempted to use it again since this issue came up so can't speak on if the earlier versions will work.