daynix/UsbDk #131: Opening the AX201 Bluetooth device causes a blue screen

ggaudeau, 2024-05-10:
libusb 1.0.26
usbdk 1.00.22

When calling the libusb_open method with the wireless controller AX201 Bluetooth, I always get a bsod WDF_VIOLATION (10d).

Reproduced with several laptops with Bluetooth driver enabled.

Workaround found for the moment is to filter all devices belonging to wireless controller class.

[windbg.txt](https://github.com/daynix/UsbDk/files/15271626/windbg.txt)

Tesquibel, 2024-05-10:
Do what

Tesquibel, 2024-05-10:
Fuck off

ybendito, 2024-05-10:
@ggaudeau 1.0.21 will not produce BSOD but may fail to open the device when it is in low power mode

ggaudeau, 2024-05-10:
Thanks for the reply @ybendito, I've just read the post [here](https://github.com/daynix/UsbDk/issues/115) which explains quite well what's going on.

Tesquibel, 2024-05-10:
We will see what my lawyer thinks about this