daynix/UsbDk #134: Installing UsbDk kills Windows 11 24H2 entirely until removed

UDPSendToFailed, 2025-02-09:
Just wanted to post this in case someone installs UsbDk on a recent Windows version and faces the same issues, so they wouldn't waste their time on unsuccessful recovery attempts for hours like I did...

I installed the latest 1.0.22 x64 version on Windows 11 24H2 26100.3037, rebooted, and the system wouldn't boot up anymore, it got stuck on the boot screen with the spinning animation forever. I tried booting into every Safe Mode option which also resulted in the same, except without the spinner, but still stuck and unable to boot even after 30 minutes of waiting. 

My next step was manually removing the driver from the Program Files and Windows folders and nuking the registry entries related to it, which finally allowed Windows to boot up to the login screen with one little issue: using USB devices basically became impossible, the system wouldn't detect a keyboard or mouse no matter the port it's plugged into. At this point, I have already nuked all the remains of UsbDk from the system, yet somehow it still made USB unusable.

Then, I booted another Windows installation from a different drive taken from my laptop and put into my PC, ran SFC and DISM on the offline system, both of which returned no errors regarding the system integrity. So, another reboot into the dead installation just to be greeted with the same problem as before; completely dead USB devices, even the backlight on my keyboard wouldn't turn on.

The final solution was using an old System Restore point, which fixed the detection of USB devices. Luckily I had it enabled, otherwise I would be stuck with an unusable Windows installation, which probably can only be fixed with a complete reset.

Secr3ts, 2025-02-17:
Seconding this. Now my work pc is gone.

viantmocy, 2025-10-20:
maybe some issue is similiar like my Windows 10 but  in my case UsbDK 1.0.22 x64 have driver Unsigned , so i decide to use disable driver enforce  for solving relating unsigned driver issues but who know when iam rebooting my pc with disable driver enforce  my usb devices is kindly like wierd , mouse didt work , keyboard not work too , either for usb plug and play devices are not working. 
forcing reboot for my windows into recovery mode  in here my keyboard and mouse work properly , so iam not expert in windows just Re installing Windows for me for solving the issue.

gunjanjambhulkar, 2026-09-03:
I can reproduce a related issue with UsbDk 1.0.22 on Windows 11. UsbDk was installed by Analog Devices CCES 3.1 and registered as UpperFilters=UsbDk under the USB class GUID {36FC9E60-C465-11CF-8056-444553540000}.

With this filter active, VirtualBox 7.1.4 and 7.2.2 could not attach multiple unrelated USB devices, including a Silicon Labs CP2102N, Jabra devices, Saleae Logic Pro and an Analog Devices USB device. The typical error was:

USB device is busy with a previous request
E_INVALIDARG (0x80070057)

After removing only the UsbDk entry from UpperFilters and rebooting, VirtualBox USB passthrough started working again for the CP2102N and Analog Devices USB device.

I also tested disabling the UsbDk service while leaving UpperFilters=UsbDk registered. After reboot, local USB keyboard and mouse stopped working. Restoring UsbDk to SYSTEM_START and rebooting restored them.