daynix/UsbDk #135: Issue Installing UsbDk on Windows 11 ARM (M1 Pro)

BASHEER-A, 2025-03-09:
Issue Installing UsbDk on Windows 11 ARM (M1 Pro)

Dear UsbDk Support Team,

I am trying to install UsbDk on Windows 11 ARM, running inside Parallels Desktop on a MacBook with an M1 Pro (ARM64) processor. However, during installation, I encounter the following errors:
	1.	“Failed to start the driver on the system, installation aborted! Please make sure you are installing a signed version of UsbDk or else try to disable driver signature enforcement.”
	2.	“There is a problem with this Windows Installer package. A program run as part of the setup did not finish as expected.”

I suspect that the issue may be due to architecture incompatibility, as I could not find an ARM64 version of UsbDk.

Could you please confirm whether UsbDk supports Windows 11 on ARM64? If not, is there any workaround to make it work in this environment?

Thank you for your support.

Best regards,

[image]
[image]

YanVugenfirer, 2025-03-10:
Hello, there is no version of USBDK for arm currently available

misha99fr, 2025-03-24:
This isnt fixable at all, fix is the arm64 version of usbdk, but usbdk doesnt have arm64 ver.