daynix/UsbDk #137: HCMON error

NullPointerExeptionBruh, 2025-05-06:
When I open the Event Viewer, I get warnings from hcmon in the system tab during system startup, with the error “Detected unrecognized USB driver (\Driver\UsbDk).”. Can this prevent the program from working?
OC: Window 10 x64 Pro
USBDk version: 1.0.22 x64

ybendito, 2025-05-06:
@NullPointerExeptionBruh HCMON is related to VMWare, Do you have VMWare installed together with UsbDk?

NullPointerExeptionBruh, 2025-05-06:
> @NullPointerExeptionBruh HCMON is related to VMWare, Do you have VMWare installed together with UsbDk? 
Yes

ybendito, 2025-05-06:
> Can this prevent the program from working?

Which program are you asking about?

NullPointerExeptionBruh, 2025-05-06:
> > Can this prevent the program from working?
> 
> Which program are you asking about?

About USBDk

ybendito, 2025-05-06:
We have no info what exactly HCMON does and what was fixed in it.
The place is "System\\CurrentControlSet\\Control\\Class\\{36FC9E60-C465-11CF-8056-444553540000}\\UpperFilters
In general both hcmon and usbdk should be listed in this MULTI_SZ registry value
If UsbDk appears first, you can try to swap them and place usbdk at the end.
But: what you see? does usbdkcontroller.exe show devices?

NullPointerExeptionBruh, 2025-05-07:
> We have no info what exactly HCMON does and what was fixed in it. The place is "System\CurrentControlSet\Control\Class\{36FC9E60-C465-11CF-8056-444553540000}\UpperFilters In general both hcmon and [...]

Where can I find "usbdkcontroller.exe"?
And in System\CurrentControlSet\Control\Class\{36FC9E60-C465-11CF-8056-444553540000}\UpperFilters I only have USBDk, there's no hcmon.

ybendito, 2025-05-07:
> Where can I find "usbdkcontroller.exe"? 
Under %programFiles%\UsbDk Runtime Library

>And in System\CurrentControlSet\Control\Class{36FC9E60-C465-11CF-8056-444553540000}\UpperFilters I only have USBDk, there's no hcmon.

what is output of "sc query hcmon"?

NullPointerExeptionBruh, 2025-05-07:
> what is output of "sc query hcmon"?
Service_Name: hcmon
Type: 1 KERNEL_DRIVER
State: 4 RUNNING
(STOPPABLE, NOT_PAUSABLE, IGNORES_SHUTDOWN)
Win32_Exit_Code: 0 (0x0)
Service_Exit_Code: 0 (0x0)
Checkpoint: 0x0
Wait_Status: 0x0
(what?)

ybendito, 2025-05-07:
also pls output of "sc qc hcmon"
What is the problem you're trying to solve or looking for root cause of?
What is your goal? Is it to run [VMWare VM with USB redirection] and simultaneously use UdbDk? Or something different?